gabonator/DS203 #32: Prebuilt binaries don't work on my device with HW v2.81

ScruffR, 2015-01-12:
I have tried all four binaries but none of them worked on my device.
I'm using Win7 Pro 32bit (Win8.1 Pro 32/64bit don't work) and always got a `.RDY` response, but when rebooting the device I always got a "Configuration Err" and had to reflash the whole lot FPGA_281, SYS_B160 and PA1_V110 to get my device running again after each attempt.

hlinden, 2015-05-16:
Same here. Pity.

tuxadmi, 2015-05-18:
Same problem....

CC1337, 2015-07-05:
Same here, GABOUI_4.hex worked but killed the original firmware, too. Would be nice if this can be fixed because GABOUI_4.hex had awesome features!

reModding, 2015-08-14:
@CC1337 Which FPGA, SYS, ... versions do you use with GABOUI_4.hex?

CC1337, 2015-08-18:
@reModding 
Hardware Ver 2.81
SYS Ver 1.64
APP (Plus A1) Ver 1.13

reModding, 2015-08-20:
Hm, I copy SYS_B164, the latest GABOUI_4.hex and the FPGA_281.ADR/.BIN, but this does not work anyway with Hardware 2.81.

gigios, 2015-08-26:
I have tried also on my 'new' dso203, HW 2.81 SYS 1.64 and APP 1.13
Also on my case I have seen many problems as "FPGA Configration Err!" message at boot phase.
The second effect is that now, when I reconnect the dso to the PC (in DFU mode), the filesystem shown 3 folders with chines chars and it is not possible to rename/delete them.
I have tried to restore the original firmware/fpga/app and all seems working fine but in DFU mode I have always the 'strange' folder structure.

Have you any idea on how restore the original filesystem? (I'm not able to change on windows 7, probably for some filename limitation that inhibit the access to the wrong files/folders).

pit001, 2015-08-27:
Hello, I also tried to get it running. 
Have also the new HW 2.81. one time I got the Firmware running, but I didn't get a input signal.

I also have the scampled directory names (like reported from gigios)

I hope there is a chance to fix it. 
Thx, Pit

Hardware Ver 2.81
SYS Ver 1.64
APP (Plus A1) Ver 1.13

gigios, 2015-08-31:
I'm not sure but seems that the project is died!
There is not any answer to any question for months.

What I have seen on the seeedstudio forum is that, probably, the incompatibility is raised from the memory management. Seems that the trick used in the app, in the HW 2.81 overwrite part of the memory used by the FPGA.

Serveurperso, 2015-11-06:
Please same problem here HW 2.81 fail

6HJS, 2016-03-24:
I am working on HW2.81 and hops to make the GABOUI works.

by modifying the \DS203-master\Source\HwLayer\ArmM3\lds\app1_win.lds

line 5 	
ROM (rx) : ORIGIN = 0x0804C000, LENGTH = 200K
to 
ROM (rx) : ORIGIN = 0x0802C000, LENGTH = 200K
to aviod overwritting FPGA field, I can got DS203 starts up.

Wildcat talked about the memory mapping issue here.
http://www.seeedstudio.com/forum/viewtopic.php?f=26&t=5512#p20206

but it gives a SDK Warning: Invalid request
and
699 ../Source/Framework/Eval.h Assertion fail.
[image]

6HJS, 2016-03-24:
Well here is my sollution, it is sample and works well.
Download the whole Github repository in zip file.
Unzip to a path contains NO blank space.
Download the ARM toolchain form [https://sourcery.mentor.com/sgpp/lite/arm/portal/package8735/public/arm-none-eabi/arm-2011.03-42-arm-none-eabi-i686-mingw32.tar.bz2](url)
And past the \arm-2011.03 to \DS203-master\Toolchain\
Edit the \DS203-master\Source\HwLayer\ArmM3\lds\app1_win.lds
line 5
ROM (rx) : ORIGIN = 0x0804C000, LENGTH = 200K
to
ROM (rx) : ORIGIN = 0x0802C800, LENGTH = 200K
Run the make.bat in \DS203-master\Build\arm_win32
Finally copy the GABOUI_1.hex form \DS203-master\Bin to your DFU driver.
Reboot the device with out pushing any key.
[image]
It runs well without warning jumps out any more.

pit001, 2016-03-24:
Hi 6HJS, good job ! I wait sooo long for an update :)
Can you correct your links an maybe upload the hex files ? Thx Peter

pit001, 2016-03-24:
Hello, tryed it and it's working !!!

Some hints: 
- The Link content to the ARM toolchain is wrong  - copy the text to your Browser
- If you use the ARM toolchain you have to change the makefile.bat (set CBASE=..\Toolchain\arm-2011.03\bin\)
- it's makefile.bat not make.bat
- in my ARM toolchain some files has zero-size. So I use the http://pub.valky.eu/arm-2011.03-lite.zip 
 (It's smaler, only one time packed and it works).

Attached - the resulting .hex file (delete the ending .txt)
[GABOUI_1.hex.txt](https://github.com/gabonator/DS203/files/187541/GABOUI_1.hex.txt)

gabonator, 2016-04-01:
6HJS, great work! Which HW versions require the linker script change? I will update the linker script and generate firmware images for both older hw and newer hw

6HJS, 2016-04-06:
@gabonator Only the hardware version with ICE40 FPGA needs to change the linker script.

The structure of FPGA image is also changed, therefore the LOGO and FPGA version in your system can not display properly.

The Chinese developer has released the FPGA source code, please have a look.

http://minidso.com/forum.php?mod=viewthread&tid=149&extra=page%3D1

joeig, 2016-05-11:
Works!! Thank you!

closed 2016-05-13 by gabonator

CC1337, 2016-08-22:
GAB_N2.hex works on my HW 2.81 as well! Thanks!