gabonator/DS203 #36: Saving Error

BobbyCephy, 2015-07-30:
Hello,
many thanks for you awesome firmware. I cant imagine how much work and time it took you to create it. Unfortunately i cant save anything on the internal storage except the settings file. Evertime i try to save something in the wave manager the ds203 freezes. After a long time it responds again normally but no files are visible. After a restart the files are appear with proper sizes but they cant be opened. Windows notifies that the files are not accessible. Also the settings cant be loaded and saved anymore. Formatting the storage makes the space usable again. So saving files destroys the filesystem guess. I have the hardware 2.72 and installed these versions: FPGA261, PA1_V113 and SYS_B162.
It would be really nice if you could help me!

thekk1, 2015-12-04:
same problem here with the same hard- and software. tested also with SYS_B164.

ferrader, 2015-12-20:
Hi, to solve the problem we need to use error checking in windows disk properties

flbauer, 2020-05-25:
In my case the flash drive got corrupt over the time. I could not save files nor calibration data.
Reflashing the FPGA-configuration and installing the original software temporarily solved my problem.

closed 2022-08-23 by BobbyCephy