gabonator/DS203 #38: Unable to save data

morfeus2, 2016-12-15:
Hi, I do not know if you (author) are still monitoring this, but I hope so.
I am not able to save *any* data (BMP, DAT, CSV): I always get file and filesystem corrupted. BMP and CSV take also very long time, in the order of 1-2 minutes.
It's really a pity because your firmware is really useful and very convenient to use. Without the possibility to save the acquired data (I am interested mainly in CSV!) an oscilloscope I would say is half-useful.

I tried all I could: changing the SYS (B160, B162), applying ALT_04.HEX, preparing dummy files to be overwritten, formatting the drive with FAT12/FAT16/FAT32.
I looked on all the possible forums, but I cannot find a solution.

Now I am with:
HW:  2.82
SYS:  1.64
FPGA: 0.00
DFU:  V3.46C

Sometimes I just bet corrupted contents and wrong file length, sometimes random files appear on the partition.
Anything that could point me to a direction to solve the problem would be really appreciated.

KaiJan57, 2017-01-26:
Same with
HW: 2.72
SYS: 1.62
FPGA: 261
DFU: V3.12C

ottelo9, 2018-04-18:
Cannot save anything:

HW: 2.72
SYS: 1.64
FPGA: 0.00 (2.61)
DFU: V3.12C