gabonator/DS203 #39: Problem with inputs

lucasgonzalez939, 2017-07-12:
Good morning, I have a DSO203 Hw version 2.60 with SYS version 1.50, yesterday I was able to finally upload the Gabonator .hex files, I used the Nx.hex files, and right after installing I couldn't get any input signal, so I thought maybe just needed to calibrate, but after the calibration he still doesn't show any input. I have installed a HC05 "linvor" bluetooth module connected to the header in the mainboard who sends to the app in android some regular signals but do not correspond to any input, one of my worse guesses is I have damaged the board in the process, although I was very careful with it, I know it's delicate hardware in many ways, another guess is I have to upload the Old .hex files instead of New ones because of Hardware compatibility maybe. In any case I would like your help, maybe you have seen a similar problem before and you are able to suggest me something. Thank you

aschwoerer, 2017-07-12:
Hey there, 
i have had the same problem a few months ago on my DSO on my first try. I have a 2.72 and tried to flash Nx.hex.
Probably this corrupted my FPGA Code. If I remember correctly I had to flash the original FPGA file from the wiki and used an Ox.hex file afterwards. Then the inputs worked correctly.
Kind regards
Andi

lucasgonzalez939, 2017-07-12:
Andi, thank you for the reply, I will try that today, and tell you how it goes, thank you so much.
Best regards.

lucasgonzalez939, 2017-07-12:
I did exactly as you said Andi, flashed the original FPGA and after rebooting the DSO flashed the Ox.hex files, after that, everything was working correctly and problem solved. Thanks again.

closed 2017-07-12 by lucasgonzalez939