gabonator/DS203 #40: FPGA Configuration Error

ottelo9, 2018-04-12:
I updated my DS203 from firmware Februar 2015 to the newest one. But now I have big problems. Logo is missing/broken, startup shows error with Configuration.cpp and FPGA Configuration errror. Measurement is not working anymore.

ottelo9, 2018-04-16:
Ok I solved it. I flashed the new (N) firmware to my V2.72, by mistake and so overwrite parts of my FPGA?! I could solve it after reflash FPGA 1.62 and SYS 1.52. 

Download is here: http://minidso.com/forum.php?mod=viewthread&tid=149&extra=page%3D1

All other Downloads on seedstudio are offline!!

Remain problem is missing logo...

ottelo9, 2018-04-18:
Ok thanks. I've now the problem that the DSO203 dont save anything ...

Mysticcccc, 2019-07-13:
Hallo,
I have the same Problem.
Original: 
DS203
HW V2.72
Sys 1.62
App 1.12
Ser: 811EC184

I have install the gabonator on Slot 4
Now I get the FPGA Configuration Error on boot
Boot Logo is broken
in gabonator Chanel1+2 no Signal

which files do I need and in which order do I have to install them?
I have tried
S203.FPGA.V2.61.ADR
DS203.LOGO.ADR
SYS_B164.hex
PA1_V113.hex
then Reboot.
no reaction, still there FPGA Configuration Error 
can you help me
how do you get the software deleted on slot 4?
it's so confused,
Many Thanks

Mysticcccc, 2019-07-26:
Hi,
cool , the problem is solved.
I flash many custom programs ( Logik Analyzer  LOGI_8MB.HEX , LOGIC03.HEX .. )
Then I flashed the FPGA Files, Sys Files and App Files.
Now the device ist ok

katze56633, 2019-12-24:
Could you write more in details what you have done?
I do habe the same problem.
Hardvare V2.70
sys ver 1.52

I installed 
fpga 2.61
but i wont help

I still get the FPGA error.

best

gabonator, 2019-12-24:
@katze56633 try flashing the FPGA v 281 from this webpage:
http://www.minidso.com/forum.php?mod=viewthread&tid=666

Refer to the Upgrade procedure described here:
http://wiki.seeedstudio.com/DSO_Quad/

If it won't work, flash whole device with USB to UART converter:
https://github.com/gabonator/LA104/tree/master/tools/stm32flash

There is also new feature available for DS203 - web usb oscilloscope:
https://github.com/gabonator/LA104#web-usb-oscilloscope

Mysticcccc, 2019-12-24:
Hi,

I flash many custom programs ( Logik Analyzer LOGI_8MB.HEX , LOGIC03.HEX .. )
Here:
https://jpa.kapsi.fi/dsoquad/

you need a large file that overwrites the storage space. I think the LOGI_8MB.HEX overwrite all old Programms and Configs.

Then I flashed the FPGA Files, Sys Files and App Files.

Good Luck

katze56633, 2019-12-24:
(History: Everything was fine, until I installed GAB_N1.hex.
Now the DSO cant measure, and I get the FPGA configuration error)
-Download DONE
-Installation done.

But no difference, 
I do have still the same problem.
FPGA configuration Error.
Question:
Fits the FPGA v281 files to HW2.70?
I read something I need FPGA 1.62 with HW2.70.

"If it won't work, flash whole device with USB to UART converter:
https://github.com/gabonator/LA104/tree/master/tools/stm32flash"
I need a 64 bit System, right? 
That I don’t have.
But I get it maybe tomorrow.
Could you explane the proceed more in details?
Which hardware so I need?
Just Win10, USB cable and the stm32flash.exe?

gabonator, 2019-12-24:
it does not matter what system (32/64bit) you have, just try to find stm32flash which will work on your machine (it can be compiled for any architecture win/linux/mac). open the device and near the power switch, there is connector named CN6. I soldered small pin header there, so I can easily flash whole device when neccessary. you need to short BT0 signal to VCC (i guess thats are the first two pins), and then connect RX, TX and GND with your usb to uart converter running on 3.3V. run stm32flash COMxx: to see if it identifies the MCU correctly

katze56633, 2019-12-25:
So,
yesterday I got my 24-12 Christmas Present.
The DSO works.

Actually, in details it is not so clear what was the right step, because I tried so many files.

I installed the LOGI_8MB.HEX , LOGIC03.HEX  files many times
Then 
DS203.FPGA.V2.61.ADR
DS203.FPGA.V2.61.BIN.
SYS_B152.hex
DS203.V1.13.hex

I think then it works, the software started.

I installed the GAB_O1.hex.

I calibrates the DSO.
The I started the Logi software, which was on application 3.
After that, the calibration on application 1 was lost, the K values where out of range.
So I had to calibrate again.

Everything works.

Some questions:

In About-Device stands FPGA- Version 0.00 (s. file attached)

What is the noise picture above?

flbauer, 2020-05-25:
After a "near brick situation" I have now a similar effect. Even the DFU version is now a smiley.
But I can live with it and am glad that the device is working again with the great software from Valky

[image]

gabonator, 2020-05-25:
This project is partially abandoned. I recommend to switch to this one:
https://github.com/gabonator/LA104/releases/tag/2019.12

It allows you to use the device as USB oscilloscope

acidg, 2020-11-18:
Any chance someone of you still has the DS203.FPGA.V2.61.ADR and DS203.FPGA.V2.61.BIN files? I can not find it anywhere on the internet and flashed FPGA 281 to my HW 2.72. Now It is bricked and doesn't show any values.

MotoMaxis, 2020-11-18:
[hwV272-V261FPGA.zip](https://github.com/gabonator/DS203/files/5563188/hwV272-V261FPGA.zip)

acidg, 2020-11-19:
Thanks a lot!

AndPot, 2022-12-03:
Hello guys, I've a big problem...get always FPGA configuration error with all sys or app flashed! The only fw that works is ds203 beta but miniosc have hardware fault.
This my specs device:
Hardware
==============
DFU version: V3.45C
Hardware version: 2.81
System version: 1.64
FPGA version: 0.23
Display: (null) (400x240)
Disk: (null) (8192kB)

Software
==============
BIOS version: 0103
Build rev: 39c042b
Build date: Nov  5 2020 15:24:49
Build user: gabrielvalky
Build system: darwin19

License
==============
Serial number: 7c2ccaac
License number: 00000000
License valid: 0

[image]
[image]

At startup I get FPGA error with all fw,app and sys of anyone developer and when try to run miniosc, hardware fault as in picture.
Thank you so munch in advanced for help...I'm very frusted, I hope to find a fix.

solarpirats, 2024-07-18:
> [image] So, at first many thanks to all helpers.
> 
> yesterday I got my 24-12 Christmas Present. The DSO works.
> 
> Actually, in details it is not so clear what was the right step, because I tried so many files.
> 
> I installed the LOGI_8MB.HEX , LOGIC03.HEX files many times Then DS203.FPGA.V2.61.ADR DS203.FPGA.V2.61.BIN. SYS_B152.hex DS203.V1.13.hex
> 
> I think then it works, the software started.
> 
> I installed the GAB_O1.hex.
> 
> I calibrates the DSO. The I started the Logi software, which was on application 3. After that, the calibration on application 1 was lost, the K values where out of range. So I had to calibrate again.
> 
> Everything works.
> 
> Some questions:
> 
> In About-Device stands FPGA- Version 0.00 (s. file attached)
> 
> What is the noise picture above?

Hi katze56633,
I know this chat is old but my last chance...
I have the same problem, but I cannot find the source files anymore, all links are dead. Do you still have the source files and can you somehow provide it on a cloud drive? Would be very helpful, thanks a lot!

flbauer, 2024-07-20:
What source files are you looking for? 
Did you check the Chinese Miniware forum?
https://forum.minidso.com/

solarpirats, 2024-07-21:
I searched for the files katze56633 mentioned, but no result. My DSO gets stuck with message "FPGA configuration OK!" HW-version 2.82, SYS version 1.64
I would like to reprogram the device, I tried all suggestions here, but nothing helped. My mandarin language skills are not existing, so the forum is not really helpful...
I got two files from support, PA1_114.hex and SYS_B164.hex, both failed to load (changed to .err)