gabonator/DS203 #42: How To Use Serial Monitor?

TSJim, 2018-09-08:
How do I use the serial Monitor feature on the Gabinator DS203 firmware?

I'm hoping that I can connect it to an Arduino which is being powered by a non-USB power source (through VIN) and be able to see messages that the Arduino sends out with Serial.print functions.

I tried connecting the probe from the DS203's Channel A port and the tip to the Arduino's TX pin and the probe's alligator clip to GND on the Arduino, but nothing shows up.  I also tried the other channels on the DS203, and tried connecting the tip to RX, GND and various combinations, but I'm not seeing any text displayed on the screen.

I am using 115200 on both devices.

Is there a trick to this, or am I just missing something?  ;)

Thanks!

gabonator, 2018-09-10:
Hello,

serial monitor just shows data received on UART RX pin of the PCB board. Unfortunately this feature is not related to the analog input channels. If you disassemble the oscilloscope, you will find there 6 pin header marked as CN7, this connector is used mainly for device recovery when you brick your device by flashing corrupted firmware. Here is a tutorial how to use hardware uart of ds203:
https://github.com/gabonator/DS203/tree/master/Man/SerialOutput <https://github.com/gabonator/DS203/tree/master/Man/SerialOutput>

There is also some very simple uart decoder for analog input channels directly in the ds203 firmware, but it cannot monitor the source continuously:

 If you need to monitor multiple UART signals at the same time (but it cannot handle 115200 bauds, I am using it for monitoring 3 lines of 9600 baud uart), you can use cheap arduino compatible board called ESP8266:

https://github.com/gabonator/Projects/blob/master/MultisampledUart/bitbang.h <https://github.com/gabonator/Projects/blob/master/MultisampledUart/bitbang.h>

or for single channel (any baudrate):

https://github.com/gabonator/Projects/blob/master/WifiUartMonitor/firmware/sender.ino <https://github.com/gabonator/Projects/blob/master/WifiUartMonitor/firmware/sender.ino>

Hope this helps,
Gabriel

TSJim, 2018-09-11:
Hi Gabriel...

Wow, I really appreciate your reply.  This is great information!  I can
hack my DS203 no problem with your links,

Thank you!

TSJim, 2018-09-17:
Hi Gabriel...

I did the modifications to my DS203 tonight thanks to your link to the
information!

However, now when I connect the TX and RX lines on the DS203 to any live
serial device (or connect the two leads together without any serial device
connected at all), I get the display on the DS203's screen such as this:

...
TEST!Ready.(290)
TEST!Ready.(291)
TEST!Ready.(292)
TEST!Ready.(293)
...

The number in the parentheses starts at "2" when the DS203 unit is turned
on, and the RX and TX leads are connected to a serial device or each other.

What is happening here?  Is the serial monitor User App working in the
newest firmware, or is it incomplete?

Thank you!

gabonator, 2018-09-18:
I am glad that it works for you :) 
The private.h should contain these variables:

const char* ssid     = "wifi-ssid";
const char* password = "wifi-password";
const char* host = "api.yourserver.com";
const char* script = "/path/to/php/script?uartmon";

I usually define default variables in main .ino file as macros, and then override them in private.h file. This time I forgot to do it in this way, but I think it is pretty straightforward to see what variables are missing and what is their meaning. In my repository, I set up the .gitignore file to ignore all "private.*" files, so I am not publishing my wifi password to public repository...

TSJim, 2018-09-19:
Hi Gabriel...

OK, thank you!  I will follow your guidance.