gabonator/DS203 #44: DS213 Compatibility #Enhancement Request

PawxSpec, 2019-01-27:
Any compatibility with the DS213? 

http://www.miniware.com.cn/product/ds213-mini-oscilloscope/

Other sketchy sources (ebay, banggood.com, etc.) say this has 15MHz Analog Bandwidth. 
From main website: (link above)

CPU: ARM Cortex M3

FIFO & Control: FPGA

ADC: HWD9288-100

Analog Channels: 2（[CH_A] [CH_B]）

Digital Channels: 2（[CH_C] [CH_D]）

Operation Channel: INV [CH_A], INV [CH_B], [CH_A]+[CH_B]、 [CH_A]-[CH_B]、[CH_C]&[CH_D]、[CH_C] | [CH_D]

Vertical Sensitivity: 10mV-10V/div (1-2-5 STEP) (using X1 probe)

                                  100mV-100V/div (1-2-5 STEP) (using X10 probe)

Vertical Resolution: 8 bits

Sample Memory Depth: 4 channels, 4*4K

Coupling: DC, AC

Max Input Voltage: ±40Vpp (using X1 probe)

                 ±400Vpp (using X10 probe)

Trigger Types: Ascend/Descend Edge Trigger Mode, Positive/Negative Pulse Trigger, Logic Trigger

Trigger Mode: Auto, Normal, Single, Slow

Trigger Source: [CH_A] [CH_B] [CH_C] [CH_D]

Signal Source: Square Wave: 10Hz to 8MHz, in total 20 shifts and 3 Vpp Signal Sources;

Sine/Triangle/Sawtooth Wave: 10Hz to 20MHz, in total 11 shifts and 3 Vpp Signal Sources

Storage: built in 8M USB disk

File Types: BMP、BUF、CSV

Auto Measurement Types: Signal sequence/cycle/duty ratio, peak voltage/virtual value/max value/min value/average

Cursor Measurement: Time value, Amplitude value

Display Modes: 4 wave lines: Line 1: [CH_A], HIDE

Line 2: [CH_B], HIDE

Line 3: [CH_C], REC_1, REC _2, REC _3, REC _4, HIDE

Line 4: [CH_D], [CH_A]+[CH_B], [CH_A]-[CH_B], [CH_C]&[CH_D], [CH_C] | [CH_D], HIDE

Sampling Mode: Real-time Sampling

Sampling Rate: 100M Sa/s

Battery: 1000mAh

Size: 99.5mmx58.5mmx13.5mm

Weight: 129g (with battery)

Certificate: CE, FCC

gabonator, 2019-01-27:
Hello! I am currently working on firmware for LA104 (https://github.com/gabonator/LA104) which is device produced by the same company and uses the same processor. I took a bit different approach - simple operating system that can load ELF application images, so the development is much faster and also you do not need to think "which memory slot contains which app?" Just nice simple graphical application switcher solves this task:

[image]

So a lot of things will be reused from LA104 development. I do not have capacity to maintain this (DS203) firmware and I noticed that every week (or so) a new issue is reported. So I am hoping that new DS203/DS213 fw will be compatible with all hardware revisions and we will solve these hiccups with choosing the right firmware image for your oscilloscope - because there were multiple hardware revisions and you need to choose the right image version to get it working. Also people were puzzled to choose the right firmware, because there were 4 different ones which loads to 4 different slots. Not everyone understood the idea that you can load 4 different firmwares onto the device.... (just look into issues tab).

So I am now waiting for e-design to provide me with all source codes to start porting this (DS203) firmware to the new design. 

Just out of curiosty, if anyone is reading this, give +1 if you also request the firmware for DS213, so I will know whether there is actually demand for it.

When I will start the development for DS213, I will let you know here

Thinstation, 2019-06-05:
+1 for DS213

eloktrobanause, 2019-06-15:
+1 for DS213

eloktrobanause, 2019-06-15:
specially x/y function to measure sinus cosinus figures

chjwch, 2019-06-15:
+1 for ds213

MightyBohab, 2019-06-22:
+1 for the DS213 for sure. I love this little thing but it needs some work.

mperci, 2019-06-28:
+1 for DS213.

jamespatten, 2019-07-07:
+1 for ds213
Thank you for working on this

ImmortalRat, 2019-07-12:
+1, keep up the great work! amazing!

cmorche, 2019-07-14:
+1 for DS213 as well! Happy to test as well. I opened it up and found an SOIC-8 SPI flash chip, would a dump of that be useful?

When I originally got this, I thought it would be compatible with these existing FW's, as I thought only the analog input rate was increased. I've been so bummed, especially because it was like three times the cost of a DS203, or even a LA104, and the existing firmware has been so difficult to use that I haven't even used it successfully once...

gabonator, 2019-07-15:
@cmorche, that SPI chip contains the FAT filesystem you see when you attach your device to PC, so it holds only configuration and possibly screenshots and caputred signals. I am currently working on compatibility with DS213 and it works quite well, at least the operating system stuff, now I just need to figure out how to read captured data from FPGA and combine it with the fw I made for DS203. I will focus mainly on DS213, I cannot promise anything regarding the DS203, since it was made in various hw revisions and for me it is absolute magic what are the differences and I just dont want to spend much time with backward compatibility. Guys at e-design will be surely very happy, when owners of DS203 will upgrade to DS213 to try new features... :)

caykroyd1, 2019-07-31:
+1 for DS213....  I would be happy to help debug when you get an initial firmware.  The current firmware is garbage and cannot trigger using single or normal trigger mode with high sample rates.  It makes the product useless to me, so I really hoping you can fix and provide a new firmware quickly.

cmorche, 2019-08-01:
@gabonator I'm not sure if it's helpful, but I just noticed they released the source code for the latest FW update for the DSO213. Looking forward to any updates :)

http://www.minidso.com/forum.php?mod=viewthread&tid=3357&extra=page%3D1%26filter%3Dtypeid%26typeid%3D154&page=1

gabonator, 2019-08-01:
@cmorche, thank you, anyway I am right now looking at this code :) I successfully ported if to my OS, but for my surprise, the trigger does not work properly. If I connect WAVE OUT to CH1 and configure trigger level and channel, the screen is not refreshing. Only when I switch to "SLOW" display mode, I can see the waveform. Strange. At first I thought it is my fault, but then I flashed their FW and it behaves in the same way.

caykroyd1, 2019-08-01:
Just to note.  The 1.22 firmware that you reference has the trigger bug.  You can refer to the forum link below:  http://minidso.com/forum.php?mod=viewthread&tid=3518&extra=page%3D1

I honestly don't have too much faith in mindso to fix as it has been months and they will not commit to a timeline.  Without that trigger functionality most of us think the DS213 is useless.  So, I think we are all hoping that gabonator can come in and save the day!

gabonator, 2019-08-01:
...pretty sad, I don't understand why they still didn't fix this issue. If only I could split into 5 persons, so I can do all the things I planned to do and go to work and still could have time to party with friends till late morning. But surely this wonderful piece of hardware deserves some more time, but it seems this will be problem of the FPGA code which was not released yet. They released only the code which is running on ARM processor, but not the code running on two other chips on the board.

caykroyd1, 2019-08-01:
If it is something you think you could fix, I will get on their forums and request the FPGA code.  Maybe they will provide it.  They claim it is open source, so they should make it available.

cmorche, 2019-08-01:
@caykroyd1 I was just thinking that as well. I've contacted other companies
in the past for GPL violations and source code requests, and was usually
successful (Except Panasonic...). I'll try emailing them/contacting
support, if you want to pester them on the forum?

I can't believe they are selling this for so much, when it has major
unresolved issues. Plus, the OG DS203 is going to be off the market soon.
I'm really starting to regret ignoring all the threads I read online before
I bought this, that all said to spend your money on something non-portable
and professional instead.

cmorche, 2019-08-02:
@gabonator/DS203 <DS203@noreply.github.com>  come on, you don't just want
to give up all those material needs and dedicate yourself to a niche
firmware replacement for a spectrum analyser, especially with the added
challenge of an unhelpful seller? (They seriously seem inept on their
forums, people are simply requesting copies of the original app.hex and
they cant even provide that...). I'm just kidding btw, thanks for your
effort in this, it's greatly appreciated!

Do you accept any donations? I'm unemployed ATM, but would happily toss
some coinage your way :)

gabonator, 2019-08-02:
The only donation I accept is invitation for couple of beers :) I am doing it in my free time because I am having lots of fun with it. The LA104 is great as arduino replacement, it has battery, display, simple controls, you can load it with tons of applications and switch between them very easily. The situation around DS213 is bit worse, you can use it "just" as oscilloscope, since it does not have any GPIO pins exposed and without proper firmware it is more or less useless.

They guys at minidso are very kind, every time I was facing any software issue and requesting portions of BIOS code (which is not open source, this is the way how they protect their devices from counterfeit copying), they always replied in few days. This is their response regarding trigger problems:

> Yes, we know this bug, and we are work on it. next version firmware will fix it.

So let's just be patient... It is summer, maybe their engineers are just on vacation

aviralpatel, 2019-11-06:
Today I discovered that single mode trigger works with time base >= 5ms and with timebase < 5ms it shows all the readings but doesn't show the waveform which means the waveform gets captured it just don't show up on the screen. Maybe this could be of some help

caykroyd1, 2019-11-07:
> Today I discovered that single mode trigger works with time base >= 5ms and with timebase < 5ms it shows all the readings but doesn't show the waveform which means the waveform gets captured it just don't show up on the screen. Maybe this could be of some help

@aviralpatel   Cool finding... It may be best for you to post your findings here:
http://minidso.com/forum.php?mod=viewthread&tid=3814&extra=page%3D1&page=2

They are not working very hard on fixing this, but this information might give them the push they need to fix the scope.

angelovAlex, 2019-11-26:
> @aviralpatel Cool finding... It may be best for you to post your findings here:
> http://minidso.com/forum.php?mod=viewthread&tid=3814&extra=page%3D1&page=2
>
> They are not working very hard on fixing [...]

Does not look like it's going to be fixed in this year. And they definatly won't realize the code for fpga until they fix it. Can somebody explain to me what is the problem?  When I got DS213 I noticed that issue and was trying to fix it. I found out that there is no problem with firmware but with fpga. I did some workaround like disabling other channels and it makes it to work a bit better. But still it's mostly useless and just stay in a box covered with dust. I never code fpga and I don't understand what the problem to write own firmware from scratch? Is it too difficult or impossible without their input? According to how firmware works I found that it mostly only sets sample rates and trigger, and on trigger reads values from fpga, sounds simple, doesn't it?

gabonator, 2019-12-02:
I have good news (email from minidso.com):

> FPGA chip has manufacturer level bug, but it can be repaired by software. We are coordinating the firmware upgrade.

ubidefeo, 2020-01-10:
good to know they at least give email feedback to someone, @gabonator 
I have sent them at least 3 messages over the past year (the forum is beyond useless even when they don't reply in Chinese), but nobody ever got back to me.
I believe it's unreasonable to expect you to jump on the train since it's a completely different beast compared to DS203, but I would expect them to at least update customers or communicate back.

I imagine they're not a large team, but products are beautifully made, and I hope they'll keep working on these toys.
Thankfully patience is not something I'm short at :D

darkonerakot, 2020-02-15:
Guys, I have 3 firmware 1.20,  1.21, 1.22, can it help?

darkonerakot, 2020-02-17:
1.20_119kb.  1.21_103kb    1.22_98kb  weird thing

angelovAlex, 2020-02-19:
> Guys, I have 3 firmware 1.20, 1.21, 1.22, can it help?

Do you have the bug on previous versions? I suppose yes, since the issue is in fpga. In anyway, can you post firmwares somewhere?

darkonerakot, 2020-02-20:
I didnt install firmware, I am at trip, have no possibilities right now, gonna do 23 feb, I dont care about device, I got refund and device still with me, i think I should spread awareness how dead device is. Accuracy is low, I dont know how can it be fixed through firmware update.sry bad eng.

geogeim, 2020-02-26:
@darkonerakot  even though the device has it's problems it would be nice to post the firmware/test results maybe we can figure out some workaround

mg3100, 2020-02-26:
I compiled the sourcecode, uploaded it to my device and now  I have a working NORMAL mode.
SINGLE mode still not working but for now that's ok , I can at least use it now.

[image]

darkonerakot, 2020-02-26:
1.20 gave me error:to big file, 1.21 already in my device, single mode is starting right after you push the button, not when a trigger points, for me there is no defference, single mode worked  better for me with 1.22 not always, but worked on trigger. Scale in pause doesnt work, device sucks. gonna sell it.

cmorche, 2020-02-26:
"device sucks. gonna sell it." - I only wish I had done so earlier... What
a waste of potential :(

darkonerakot, 2020-02-26:
forgot to say, despite I bought it 07.19 AliExpr returned my money back, I wrote to support and they solved this problem, I gave up on this device

sibradzic, 2020-02-27:
@darkonerakot can you please share the firmware binaries?

darkonerakot, 2020-02-27:
[https://drive.google.com/folderview?id=1BUJKbNZfNvm2rnVBeq589OYcHreeooDG](url)

gabonator, 2020-02-27:
@darkonerakot thank you for providing the firmware files, I am dropping copy of them here:

[DS213AppV121V122.zip](https://github.com/gabonator/DS203/files/4261749/DS213AppV121V122.zip)

darkonerakot, 2020-02-27:
ppl say 1.20 have no bugs with scale in pause mode, I understand that reverse engineering is pretty big deal, but did anybody see something in it? and if get this right that is not firmware, right? this are apps, some interesting thing should be in the bootloader I guess, I am not a programmer so sorry for my silly conclusions. And I think there is a lot of things in bootloader such as licence and hardware support.  correct me if wrong.

gabonator, 2020-02-27:
@darkonerakot so which of the firmwares should we try? The V121 and V122 files you provided look like application HEX files which could be flashed using DFU. And V120 look like full flash memory dump including bios, application and FGPA image and this kind of images can be only flashed using SWD interface (StLink V2 attached to USB connector)... Is there anyone who has tried this?

darkonerakot, 2020-02-27:
> @darkonerakot so which of the firmwares should we try? The V121 and V122 files you provided look like application HEX files which could be flashed using DFU. And V120 look like full flash memory [...]
I only  tried 1.22 and 1.21 as for me 1.22 is better but both of it is crap.
I would try flashing 1.20 via SWD but I have no STlink-flasher and any experience of using it. Maybe there is a method to make backup full dump before the procedure. I only have rs232ttl.  I compared 1.20.hex  and 1.21.hex it have nothing common except first line. But 1.21 and 1.22  have 422 common lines.

gabonator, 2020-02-27:
That is a good point with the license issue... I will make a application that will show the license/activation number, so you just need to note it down before full flashing... just FYI: for the DS203 I have "cracked" bios image which ignores the license verification (reverse engineered by jpa)

darkonerakot, 2020-02-27:
I think now we have something. I would try full flash.

gabonator, 2020-02-28:
Argh!! I am so stupid, I have just realized that the device shows the serial number and license number in DFU mode screen. Anyway, here is an application which shows some basic information about the device and dumps everything info file on EEPROM disk (so even flashing of the MCU, the file stays there)

Dump looks like this:
```
Hardware
==============
DFU version: V3.72C
Hardware version: SCH_V2.0
System version: DS213
FPGA version: AGL1K
Display: ILI9327 (400x240)
Disk: 8MB (8188kB)

Software
==============
BIOS version: 0102
Build rev: 8e696f9
Build date: Feb 28 2020 01:06:10
Build user: gabrielvalky
Build system: darwin19

License
==============
Serial number: 791caa9b
License number: bd93b316
License valid: 1
```

[ds213deviceinfo.zip](https://github.com/gabonator/DS203/files/4264730/ds213deviceinfo.zip)
system_ds213.hex - the file to flash in DFU mode
shell.elf - file manager
61devin.elf - device info app

gabonator, 2020-02-28:
I was able to connect the ST Link V2 to the DS213, dumped the flash info file without any problem, but it fails when I want to upload the file you provided... It is not problem with the file, just the upload fails...
```
sudo openocd -f /usr/local/share/openocd/scripts/interface/stlink.cfg -f /usr/local/share/openocd/scripts/target/stm32f1x.cfg

Open On-Chip Debugger 0.10.0+dev-00957-g9de7d9c8 (2019-11-09-23:50)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : clock speed 1000 kHz
Info : STLINK V2J29S7 (API v2) VID:PID 0483:3748
Info : Target voltage: 3.276064
Info : stm32f1x.cpu: hardware has 6 breakpoints, 4 watchpoints
Info : Listening on port 3333 for gdb connections
Info : accepting 'telnet' connection on tcp/4444
target halted due to debug-request, current mode: Thread
xPSR: 0x01000000 pc: 0x08005774 msp: 0x20005df8
Error: jtag status contains invalid mode value - communication failure
Polling target stm32f1x.cpu failed, trying to reexamine
Examination failed, GDB will be halted. Polling again in 100ms
Info : Previous state query failed, trying to reconnect
Error: jtag status contains invalid mode value - communication failure
Polling target stm32f1x.cpu failed, trying to reexamine
Examination failed, GDB will be halted. Polling again in 300ms
Info : Previous state query failed, trying to reconnect
```

This guy was facing the same problem, I will investigate it deeper during the weekend: https://www.jsykora.info/2014/10/error-jtag-status-contains-invalid-mode-value-communication-failure-solved/

darkonerakot, 2020-02-28:
> I was able to connect the ST Link V2 to the DS213, dumped the flash info file without any problem, but it fails when I want to upload the file you provided... It is not problem with the file, just [...]

Did you make full dump of current state of your device? It could help to compare with our 1.20.hex

gabonator, 2020-02-28:
I made it! And the FPGA images are different. Seems the oscillscope is working well - the image of sinus signal is perfectly stable, but I did not try all the timebase options. Now I am just thinking how to make the downgrade available for other without stlink adapter

# Flashing DS213:

## Requirements:
- ST Link V2 adapter (search for "ST-Link V2 stlink mini STM8STM32" on ebay or alibaba, it costs about $2)
- OpenOcd installed (http://openocd.org/getting-openocd/)
- Micro usb to ST Link cable - DS213 has SWD interface on USB port (black -> GND, green -> SWDIO, white -> SWCLK)

## Flashing procedue:

- At first we need to convert the flash dump (DS213 APP V.20.hex) info binary format

```
arm-none-eabi-objcopy --input-target=ihex --output-target=binary "DS213 APP V.20.hex" ds213v20.bin
```

- Install and run openocd

```
openocd -f /usr/local/share/openocd/scripts/interface/stlink.cfg -f /usr/local/share/openocd/scripts/target/stm32f1x.cfg
```

- It should respond with this:

```
Open On-Chip Debugger 0.10.0+dev-00957-g9de7d9c8 (2019-11-09-23:50)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : clock speed 1000 kHz
Info : STLINK V2J29S7 (API v2) VID:PID 0483:3748
Info : Target voltage: 3.271713
Info : stm32f1x.cpu: hardware has 6 breakpoints, 4 watchpoints
Info : Listening on port 3333 for gdb connections
```

- open new terminal window and open port 4444 (use telnet or screen, or putty)

```
telnet localhost 444
```

```
Trying ::1...
telnet: connect to address ::1: Connection refused
Trying 127.0.0.1...
Connected to localhost.
Escape character is '^]'.
Open On-Chip Debugger
> halt
> flash banks
#0 : stm32f1x.flash (stm32f1x) at 0x08000000, size 0x00000000, buswidth 0, chipwidth 0

> flash info 0
device id = 0x10036414
flash size = 512kbytes
#0 : stm32f1x at 0x08000000, size 0x00080000, buswidth 0, chipwidth 0
	#  0: 0x00000000 (0x1000 4kB) not protected
	#  1: 0x00001000 (0x1000 4kB) not protected
	#  2: 0x00002000 (0x1000 4kB) not protected
	#  3: 0x00003000 (0x1000 4kB) not protected
	#  4: 0x00004000 (0x1000 4kB) not protected
	#  5: 0x00005000 (0x1000 4kB) not protected
	#  6: 0x00006000 (0x1000 4kB) not protected
	#  7: 0x00007000 (0x1000 4kB) not protected
	#  8: 0x00008000 (0x1000 4kB) not protected
	#  9: 0x00009000 (0x1000 4kB) not protected
	# 10: 0x0000a000 (0x1000 4kB) not protected
	# 11: 0x0000b000 (0x1000 4kB) not protected
	# 12: 0x0000c000 (0x1000 4kB) not protected
	# 13: 0x0000d000 (0x1000 4kB) not protected
	# 14: 0x0000e000 (0x1000 4kB) not protected
	# 15: 0x0000f000 (0x1000 4kB) not protected
	# 16: 0x00010000 (0x1000 4kB) not protected
	# 17: 0x00011000 (0x1000 4kB) not protected
	# 18: 0x00012000 (0x1000 4kB) not protected
	# 19: 0x00013000 (0x1000 4kB) not protected
	# 20: 0x00014000 (0x1000 4kB) not protected
	# 21: 0x00015000 (0x1000 4kB) not protected
	# 22: 0x00016000 (0x1000 4kB) not protected
	# 23: 0x00017000 (0x1000 4kB) not protected
	# 24: 0x00018000 (0x1000 4kB) not protected
	# 25: 0x00019000 (0x1000 4kB) not protected
	# 26: 0x0001a000 (0x1000 4kB) not protected
	# 27: 0x0001b000 (0x1000 4kB) not protected
	# 28: 0x0001c000 (0x1000 4kB) not protected
	# 29: 0x0001d000 (0x1000 4kB) not protected
	# 30: 0x0001e000 (0x1000 4kB) not protected
	# 31: 0x0001f000 (0x61000 388kB) not protected
STM32F10x (High Density) - Rev: Y

> flash erase_sector 0 0 31      
erased sectors 0 through 31 on flash bank 0 in 0.798275s

> flash write_bank 0 ds213v20.bin
wrote 524288 bytes from file ds213v20.bin to flash bank 0 at offset 0x00000000 in 15.454797s (33.129 KiB/s)
```

## Known issues

- The device is fully working and only shows "System licence error!" message during startup
- Flashing fails in middle of operation - the openocd is not correctly configured, I just start and stop debug session in Microsoft Visual Code with "Cortex-Debug" extension before flashing

```
> flash write_bank 0 ds213v20.bin
flash write algorithm aborted by target
flash write failed at address 0x8010002
flash memory not erased before writing
error writing to flash at address 0x08000000 at offset 0x00000000
```

## Future ideas:
- Extract FPGA image and make application which will overwrite it without the need to use st link interface.

## Credits:
- darkonerakot

darkonerakot, 2020-02-29:
Anybody else tried it? I would like to know about changing scale at pause moment,  I have no stlink and it is not available in my town(

MotoMaxis, 2020-02-29:
If DS213 similar to my DS203, then everything except the DFU(bootloader) can be flashed through virtual drive. What will require is making a separate file for FPGA and APP.  I remember for FPGA I had to first copy .ADR file then .bin file. The App as a hex.

gabonator, 2020-02-29:
@MotoMaxis I tried, but for some strange reason it uploads the file to wrong offset - I compared full dump of flash after flashing the fpga through DFU. Be careful with these files, if you corrupt the FPGA image, the device will stay in DFU mode and wont run the application.

[fpga.zip](https://github.com/gabonator/DS203/files/4270714/fpga.zip)

MotoMaxis, 2020-02-29:
Just confirming. It doesn't write to 0x08070000 in ADR? If that's the case then they goofed up the DFU.

MotoMaxis, 2020-02-29:
I am not able to look at the fpga bin you provided but it seems really big. On DS203 it was something around 135kb. Perhaps something else was included from lib or app sections?

gabonator, 2020-02-29:
i dumped the memory region starting at the fpga section and ending at the logo screen (addresses are obtained from BIOS and easily verifiable by looking at the dump with hex viewer). so there is big section full of FFs

MotoMaxis, 2020-03-01:
Yes looking at Bios code it does look like 0x08070000 is start of FPGA. Have you tried flashing with a hex file?
[fpga-hex.zip](https://github.com/gabonator/DS203/files/4271268/fpga-hex.zip)

darkonerakot, 2020-03-01:
So it should work out if flash fgpa.hex then app1.20.hex?

MotoMaxis, 2020-03-01:
I believe so. I don't have ds213 so didn't try it myself.

gabonator, 2020-03-01:
DS213 has these attributes in its bios structure:
App1Base 0x08008000
App2Base 0x08028000
FpgaBase 0x08070000
LogoBase 0x0807d800
MaxLimit 0x08080000

So I extracted the fpga by this command:
```dd if=ds213v20.bin bs=1 count=$((0xd800)) skip=$((0x70000)) of=ds213v20_fpga.bin```

and converted to hex:
```arm-none-eabi-objcopy --input-target=binary --output-target=ihex --change-addresses 0x08070000 ds213v20_fpga.bin ds213v20_fpga.hex ```

When tried to upload the hex file through DFU, the upload process is supiciously fast, so it seems the dfu does not write anything (cant remember if the dfu renames it to ERR or not, will verify later)

Now I am trying to upload the fpga binary through my OS, I just need to convert the binary file into ELF somehow

darkonerakot, 2020-03-01:
I guess Dfu bootloader should be same since 1.21 1.22 doesnt contain it.

MotoMaxis, 2020-03-01:
gabonator, does your hex have : 020000040807EB on the first line?

gabonator, 2020-03-01:
@MotoMaxis
```
:020000040807EB
:10000000001200100000FFFFA20000200000C72027
:1000100052942508425294A129425294A5094210B3
:1000200084210842108421094A000000A200002116
:1000300000006020DD2000200008E240000FDF808B
...
```

MotoMaxis, 2020-03-01:
That looks good. I am really shocked that they don't allow to flash FPGA. It must be fault in DPU.

[233 more replies]