gabonator/DS203 #47: DS212 Compatibility request

gbraad, 2020-03-13:
The DS212 hasn't seen an update since ever... basically since release. The hardware is lilkely similar, if not same as mlst of the other Miniware DSOs. It is offered here...

What can be done to help/assist to get this device up to date?

gbraad, 2020-03-13:
https://github.com/kortschak/ds212

raenye, 2020-05-22:
DS212 has no FPGA, and MCU is STM32F303, instead of STM32F103 as in DS203/DS213/LA104.
But maybe some apps not using FPGA are still possible.

gabonator, 2020-05-22:
Could be possibly used as low end usb oscilloscope, or signal generator (DS203/DS213 generates the signal using internal DAC in STM32). But I think DS212 is discontinued and there is no big motivation to expand support for this device. If there is a C++ programmer who owns DS212 and wishes to port the LA104 operating system onto it, I am ready to provide support. Currently it would give more sense to spend time on DS213 compatibility, since the DS203 is discontinued as well (the FPGA chip is not produced anymore)

gbraad, 2020-05-22:
> discontinued

This is not the case. http://www.miniware.com.cn/product/ds212-mini-oscilloscope-digital-oscilloscope/ I am willing to look into this, but would like to know what would be the best way to start with this; build instructions, etc, first file to touch/look at, etc. but also, how best to approach debugging.

gabonator, 2020-05-22:
Start with this project:
https://github.com/gabonator/LA104
You need to build OS host: https://github.com/gabonator/LA104/tree/master/system/os_host
Llook inside the building scripts, there is download link to arm toolchain. Try building with "build_ds203.sh" If you are using windows platform, you need to rewrite it into batch file. It should be easy, the script is short

Then create folder ds212 here:
https://github.com/gabonator/LA104/tree/master/system/os_host/source/bios
and implement "imports.c", take inspiration from DS203 which should be most similar to DS212 (https://github.com/gabonator/LA104/blob/master/system/os_host/source/bios/ds203/imports.c)

Basically - you only need to implement pixel drawing functions, eeprom access and access to buttons and switches and the system should run on any device based on STM32F103. 

0. download arm toolchain (the only thing you need for building the os and applications)
1. try building LA104 system in os_host folder (to verify if your toolchain works)
2. implement your lcd/eeprom/button functions
3. try building shell (file manager - https://github.com/gabonator/LA104/tree/master/system/apps_shell/test29_fileman)
4. upload it to your device
5. implement ADC functions and try building the oscilloscope application (https://github.com/gabonator/LA104/tree/master/system/apps_ds213/55_gabosc)